sparks-baird/self-driving-lab-demo #9: Potential for non-linear correlations between parameters?

sgbaird, 2022-07-03:
"Mixing" usually implies first-order, linear relationships. What about higher-order correlations between input parameters? Is there a way this could be introduced physically or artificially into the setup?

I need to think about this one more.

sgbaird, 2022-07-09:
The more I think about it, the less it seems likely that this can be achieved in a low-cost setup. There's an option of mounting the LEDs to a [mini pan-tilt camera](https://www.adafruit.com/product/1967) (which introduces some very basic robotics), which would have some interesting effects. While this might introduce non-linearity, it could still probably be easily solved by gradient descent (i.e. single optimum).

sgbaird, 2022-07-09:
Independently controlling the voltage and current of individual LEDs rather than using a prebuilt library that sets brightness might be an option. Of course, would require reasonable limits so the hardware doesn't get fried easily. More work. More custom.

sgbaird, 2022-08-18:
LED current can be controlled within Blinkt!. The voltage might be fixed to a certain value https://github.com/adafruit/Adafruit_CircuitPython_AS7341/issues/1

BAMcvoelker, 2022-08-29:
Hi Sterling, 

you could perform a multi-objective optimization, where on the one hand you try to optimize the underlying physical function, and on the other hand you try to match it with known constructions that have these properties, i.e. you try to find out which setup (different "light bulbs", different building, etc.) produces this spectrum. 

You could then prioritize cheap solutions, environmentally friendly solutions, etc. to find systems that have virtually the same properties but are, for example, more ecological or economical.... 
Advantage: Depending on assumptions in costs and, e.g. CO2 balance can be any number and any complex scenarios. Does this makes sense?

sgbaird, 2022-08-29:
I like the idea of having a stochastic, physically determined objective and a deterministic one like cost. This has been on my mind, especially with e.g. LED power.

> The closer this is to a real engineering problem, the better.

Definitely agreed! Thanks for weighing in on this.

BAMcvoelker, 2022-08-29:
Right? 

We are currently building a demonstrator in which we simulate a shock in the supply chain for alloys used in turbine blade printing. This is to show the improved resilience of a data-driven R&D lab. All we need to do is change the delivery times or costs and the deck of possible solutions will be completely reshuffled.I will post a link as soon as it goes public.

sgbaird, 2022-08-29:
> I will post a link as soon as it goes public.

Please do! Sounds very interesting