sparks-baird/self-driving-lab-demo #13: Alternative demo ideas

sgbaird, 2022-07-10:
In a similar vein to a Rubens tube, vibrating a drumhead or a liquid surface and trying to match particular harmonics using speakers as inputs (a 2D signal example).

In a 3D signal example, measuring WiFi strength at 3D locations (maybe with separate sensors) where the WiFi source positions and/or sensor positions can be adjusted, and trying to match a target distribution of the WiFi signal. Starts to become reminiscent of DFT relaxation.

In a materials example, perhaps liquid suspensions of particles that are being stirred, mixing the solutions, and measuring the packing fraction (visually?) of a dried solution based on drying temperature and composition? Wash out the container with vigorous mixing and then extract the waste into a waste container (all via peristaltic pumps most likely).

Might want to look at other low-cost sensors for chemistry, physics, and materials science that might not appear on Adafruit. pH is a nice idea one that appeared in one of the papers.

sgbaird, 2022-07-10:
@Gadgetoid, came across your profile, and very nice work with pinout.xyz.

I figured it might be worth asking, any thoughts on this project? I'm particularly interested to hear your thoughts on alternative demos (hence tagging you on this thread), modifications that you think would be interesting, and suggestions on how to maximize impact.

BAMcvoelker, 2022-08-29:
In terms of a cheap sensor setup, you could use an infrared camera to validate/monitor reactivity of substances in terms of exothermic heat -> Optimize towards most even reaction or highest peak, reaction duration etc. 

Another one would be the stirring plate voltage/current to measure rheology.

sgbaird, 2022-08-29:
@BAMcvoelker great suggestions!