sparks-baird/self-driving-lab-demo #23: Suggestions for optimization algorithms to test

sgbaird, 2022-08-22:
## design of experiments (DOE)
(suggestion by @kjelljorner, see [twitter post](https://twitter.com/kjelljorner/status/1561318194786275330))
- Latin Hypercube is a common one (implemented in Olympus, see below). See also [DOEPY](https://github.com/tirthajyoti/doepy)
- [MINITAB summary of a few different DOE methods](https://support.minitab.com/en-us/engage/help-and-how-to/tools/forms-and-analysis-captures/analysis-capture-tools/doe-analysis/)
- [DESICE](https://www.desice.io/) and [JMP](https://www.jmp.com/en_us/home.html)

## L-BGFS
(suggestion by @CompRhys, see [twitter post](https://twitter.com/RhysGoodall/status/1561361955121549313))
- can use Olympus (see below), see also [scipy's L-BGFS-B](https://docs.scipy.org/doc/scipy/reference/optimize.minimize-lbfgsb.html)

## derivative-free method, e.g. simplex
(suggestion by @WardLT, see [twitter post](https://twitter.com/WardLT2/status/1561396194298933249))
- can use Olympus (see below), see also [scipy's linprog](https://docs.scipy.org/doc/scipy/reference/optimize.linprog-simplex.html)

## Olympus
Supports various optimization algorithms in addition to benchmarks (suggestion by Alan Aspuru-Guzik, see [twitter post](https://twitter.com/A_Aspuru_Guzik/status/1561436388234563585), cc @rileyhickman). Includes many options from above.

### Supported algorithms in Olympus

From ([Colab](https://colab.research.google.com/github/aspuru-guzik-group/olympus/blob/master/olympus_get_started.ipynb)) notebook

```python
['BasinHopping',
 'Cma',
 'ConjugateGradient',
 'DifferentialEvolution',
 'Genetic',
 'Gpyopt',
 'Grid',
 'Hyperopt',
 'LatinHypercube',
 'Lbfgs',
 'ParticleSwarms',
 'Phoenics',
 'RandomSearch',
 'Simplex',
 'Slsqp',
 'Snobfit',
 'Sobol',
 'SteepestDescent']
```

Olympus also has [a nice page categorizing the algorithms into "Bayesian", "Evolutionary", "Gradient", "Grid-like", and "Other"](https://aspuru-guzik-group.github.io/olympus/classes/planners/index.html). This is right in line with a results figure I'm thinking about that using plotly legend groups (see [group click toggle behavior](https://plotly.com/python/legend/#group-click-toggle-behavior)).

WardLT, 2022-08-22:
A quick note: The "simplex" I was referring to is this one - https://docs.scipy.org/doc/scipy/reference/optimize.minimize-neldermead.html#optimize-minimize-neldermead. It could be the same as the "Simplex" mentioned in Olympus. 

This simplex is a local optimizer, but you can make it "global" by just running it with different starting guesses (i.e., 'multi-start' optimization).

sgbaird, 2022-08-22:
@WardLT thanks for clarifying! Looks like [Olympus also uses Nelder Mead via `scipy`](https://aspuru-guzik-group.github.io/olympus/_modules/olympus/planners/planner_simplex/wrapper_simplex.html#Simplex). Good point about multi-start.

kjelljorner, 2022-08-22:
@sgbaird Not really an expert on DoE myself. People seemed to use proprietary software like [MODDE](https://www.sartorius.com/en/products/process-analytical-technology/data-analytics-software/doe-software/modde) and also in an iterative manner depending on how many experiments could be run in a batch etc. I'm not sure if it is possible to reproduce that workflow with open-source tools and how easy it would be to automate.

sgbaird, 2022-08-26:
I think Latin Hypercube Sampling might be a good option for DOE. Might try to use one of the proprietary softwares as well.

A visualization that I'm interested in is grouping the algorithms into categories (as mentioned above) and making it an interactive Plotly figure with [legend groups](https://plotly.com/python/legend/) and the ability to [toggle individual traces](https://github.com/plotly/plotly.js/issues/3135).