sparks-baird/self-driving-lab-demo #33: [Pico W micropython] Bayesian optimization from scratch

sgbaird, 2022-09-02:
https://machinelearningmastery.com/what-is-bayesian-optimization/ except with Gaussian Processes from scratch also

@kir486680

kir486680, 2022-09-02:
I can make a quick jupyter notebook for this, and you can take a look at it.

sgbaird, 2022-09-10:
@kir486680 do you have a notebook for this?

sgbaird, 2022-09-21:
One of the following resources is probably useful for GPR from scratch:
- https://github.com/sp8rks/MaterialsInformatics/blob/main/worked_examples/gaussian_process/coing_gpr_from_scratch_notebook.ipynb ([original source of notebook](https://gitlab.com/youtube-optimization-geeks/uncertainty-quantification/-/blob/main/coing_gpr_from_scratch_notebook.ipynb))
- https://peterroelants.github.io/posts/gaussian-process-tutorial/
- https://towardsdatascience.com/implement-a-gaussian-process-from-scratch-2a074a470bce

sgbaird, 2022-10-05:
For now, sticking with the current solution due to complexity (100's lines) of from-scratch GPR + Bayesian Optimization

closed 2022-10-05 by sgbaird