sparks-baird/self-driving-lab-demo #93: Having trouble getting the ArduCam SPI (MP2) camera to work

sgbaird, 2022-10-08:
I should probably try adding 4.7kOhm pull-up resistors

- https://stackoverflow.com/questions/73294126/i2c-not-working-on-the-raspberry-pi-pico-w-with-micropython/73359147?noredirect=1#comment130657256_73359147

[image]

- https://github.com/ArduCAM/PICO_SPI_CAM/issues/9
- https://github.com/namato/micropython-ov2640
  - https://github.com/namato/micropython-ov2640/issues/8
- https://github.com/ArduCAM/PICO_SPI_CAM/issues/8
- https://github.com/lemariva/micropython-camera-driver/issues/74
- https://github.com/edwardwebster/Micropython-Arducam/issues/2
- https://github.com/edwardwebster/Micropython-Arducam/issues/1

sgbaird, 2022-11-03:
Giving up on Arducam's OV2640 2MP. I had to replace the ribbon cable, there hasn't been any support from Arducam (forums and contact form) after about a month. The forum post I made won't even post because it's being "held for review" (~26 days). I'm planning to hold out for the [Enviro Camera (Pico W onboard)](https://shop.pimoroni.com/products/enviro-camera?variant=40056602067027) and probably set it up as its own MQTT device rather than try to do hardwired data transfer between it and the Pico W running the experiments.

Doesn't seem that I needed pull-up resistors. Pico H and Pico W seemed to get the same errors with the bad ribbon cable, both ran successfully in CircuitPython, and both had the same errors IIRC from the MicroPython implementation. Not worth it for me, hence planning to wait for something better or go with something other than the Pico. Technically I could also probably go with Adafruit's [guide on taking camera images and transferring them to the Pico](https://learn.adafruit.com/capturing-camera-images-with-circuitpython/raspberry-pi-pico-wiring); however, this is also a CircuitPython implementation.

Switching back to CircuitPython seems like a possibility now that [a bug affecting `adafruit_minimqtt` has been fixed in the firmware](https://github.com/adafruit/circuitpython/issues/6988#issuecomment-1292289775).

closed 2022-11-03 by sgbaird

sgbaird, 2022-11-03:
The [Grove Mini Camera](https://www.seeedstudio.com/Grove-Mini-Camera-p-1578.html) seems like a nice option if it was in stock and assuming it could be easily set up to work with Pico W running MicroPython.

See also:
- https://wiki.seeedstudio.com/Grove-Serial_Camera_Kit/
- https://www.seeedstudio.com/Grove-Vision-AI-Module-p-5457.html (not sure if this would have MicroPython support)

Not a color camera, but an IR camera might be interesting for other applications https://wiki.seeedstudio.com/Grove-Thermal-Imaging-Camera-IR-Array/

felixmosh, 2023-08-21:
Hi @sgbaird did you managed to connect & take picture with OV2640 & pi pico W?

sgbaird, 2023-08-21:
No, but if I come back to this I plan to follow the tutorial from https://learn.adafruit.com/adafruit-ov5640-camera-breakout/raspberry-pi-pico-usage using their Adafruit camera.