sparks-baird/self-driving-lab-demo #164: Install on MacOS Ventura with Apple Silicon

ianfoster, 2023-01-07:
Has anyone run the demo code on a Mac with Apple silicon?

Trying conda env create -f environment.yml on a MacBook Pro, MacOS 13.1, Apple M1 Max, 
I see the error shown in the attached [trace.txt](https://github.com/sparks-baird/self-driving-lab-demo/files/10364759/trace.txt), which some Googling suggests may be an Apple silicon problem.

error: legacy-install-failure

× Encountered error while trying to install package.
╰─> RPi.GPIO

sgbaird, 2023-01-07:
Hi @ianfoster, really glad to see that you got the demo up and running! The problem is likely with one of the following two:
https://github.com/sparks-baird/self-driving-lab-demo/blob/50160d67742e13c052982e743a3a2bbeca011af3/environment.yml#L43-L44

Incidentally, both are no longer needed after the switch from the Raspberry Pi 4B single board computer to the Pico W microcontroller, so I'm removing them in https://github.com/sparks-baird/self-driving-lab-demo/pull/165. If there are any other dependency conflicts you run into, it might be worth also trying to run `pip install -e .` inside of a fresh conda environment and while inside the `self-driving-lab-demo` main directory in case one of the Anaconda packages is causing an issue. The conda environment file contains

ianfoster, 2023-01-07:
That works, thanks! (And I have the Google Colab running now also, via the other issue.)

closed 2023-01-07 by ianfoster

sgbaird, 2023-01-07:
For sure! Glad it's working, and thanks for your patience. I appreciate the detailed issues.