sparks-baird/self-driving-lab-demo #167: 4.2-paho-mqtt-colab-sdl-demo-test.ipynb error?

ianfoster, 2023-01-07:
I'll try to debug this myself, but in case someone knows the answer:

Running in 4.2-paho-mqtt-colab-sdl-demo-test.ipynb on Colab the following simplified code, I get the error shown below ("ValueError: Raw data is expected to be just for one arm.")

```
from self_driving_lab_demo.utils.search import (
    ax_bayesian_optimization,
)

num_iter = 10

best_parameters, values, experiment, model = ax_bayesian_optimization(sdl, num_iter)
```

```
[INFO 01-07 17:24:27] ax.modelbridge.dispatch_utils: Using Bayesian optimization since there are more ordered parameters than there are categories for the unordered categorical parameters.
[INFO 01-07 17:24:27] ax.modelbridge.dispatch_utils: Calculating the number of remaining initialization trials based on num_initialization_trials=None max_initialization_trials=None num_tunable_parameters=3 num_trials=None use_batch_trials=False
[INFO 01-07 17:24:27] ax.modelbridge.dispatch_utils: calculated num_initialization_trials=6
[INFO 01-07 17:24:27] ax.modelbridge.dispatch_utils: num_completed_initialization_trials=0 num_remaining_initialization_trials=6
[INFO 01-07 17:24:27] ax.modelbridge.dispatch_utils: Using Bayesian Optimization generation strategy: GenerationStrategy(name='Sobol+GPEI', steps=[Sobol for 6 trials, GPEI for subsequent trials]). Iterations after 6 will take longer to generate due to model-fitting.
[INFO 01-07 17:24:27] ax.service.managed_loop: Started full optimization with 10 steps.
[INFO 01-07 17:24:27] ax.service.managed_loop: Running optimization trial 1...
[ERROR 01-07 17:24:37] ax.service.managed_loop: Encountered exception during optimization: 
Traceback (most recent call last):
  File "/usr/local/lib/python3.8/dist-packages/ax/service/managed_loop.py", line 227, in full_run
    self.run_trial()
  File "/usr/local/lib/python3.8/dist-packages/ax/utils/common/executils.py", line 161, in actual_wrapper
    return func(*args, **kwargs)
  File "/usr/local/lib/python3.8/dist-packages/ax/service/managed_loop.py", line 204, in run_trial
    _, data = InstantiationBase.data_and_evaluations_from_raw_data(
  File "/usr/local/lib/python3.8/dist-packages/ax/service/utils/instantiation.py", line 970, in data_and_evaluations_from_raw_data
    evaluations = {
  File "/usr/local/lib/python3.8/dist-packages/ax/service/utils/instantiation.py", line 971, in <dictcomp>
    arm_name: cls.raw_data_to_evaluation(
  File "/usr/local/lib/python3.8/dist-packages/ax/service/utils/instantiation.py", line 908, in raw_data_to_evaluation
    raise ValueError("Raw data is expected to be just for one arm.")
ValueError: Raw data is expected to be just for one arm.
---------------------------------------------------------------------------
ValueError                                Traceback (most recent call last)
<timed exec> in <module>

[/usr/local/lib/python3.8/dist-packages/self_driving_lab_demo/utils/search.py](https://92s1rnlmebf-496ff2e9c6d22116-0-colab.googleusercontent.com/outputframe.html?vrz=colab-20230105-060053-RC00_499751356#) in ax_bayesian_optimization(sdl, num_iter, objective_name)
     51     parameters = [dict(name=nm, type="range", bounds=bnd) for nm, bnd in bounds.items()]
     52 
---> 53     best_parameters, values, experiment, model = optimize(
     54         parameters=parameters,
     55         evaluation_function=evaluation_function,

3 frames
[/usr/local/lib/python3.8/dist-packages/ax/service/utils/best_point.py](https://92s1rnlmebf-496ff2e9c6d22116-0-colab.googleusercontent.com/outputframe.html?vrz=colab-20230105-060053-RC00_499751356#) in get_best_raw_objective_point_with_trial_index(experiment, optimization_config, trial_indices)
     94     dat = experiment.lookup_data(trial_indices=trial_indices)
     95     if dat.df.empty:
---> 96         raise ValueError("Cannot identify best point if experiment contains no data.")
     97     objective = optimization_config.objective
     98     feasible_df = _filter_feasible_rows(

ValueError: Cannot identify best point if experiment contains no data.
```

sgbaird, 2023-01-07:
Hi @ianfoster, thanks for the detailed info! The fix was on the `main` branch but not in the latest Python package version. I've updated the Python package to `v0.8.0` and verified that it runs on Colab. Lmk if that works for you.

ianfoster, 2023-01-07:
That works, thanks!

closed 2023-01-07 by ianfoster