sparks-baird/self-driving-lab-demo #171: Installing blinkt on MacOS

ianfoster, 2023-01-09:
Apple Silicon, MacOS 13.1 ...

(gtk3) /Users/ian/AAA/Code/self-driving-lab-demo/src $ pip install blinkt
Collecting blinkt
  Using cached blinkt-0.1.2-py3-none-any.whl (6.6 kB)
Collecting RPi.GPIO
  Using cached RPi.GPIO-0.7.1.tar.gz (29 kB)
  Preparing metadata (setup.py) ... done
Building wheels for collected packages: RPi.GPIO
  Building wheel for RPi.GPIO (setup.py) ... error
  error: subprocess-exited-with-error
  
  × python setup.py bdist_wheel did not run successfully.
  │ exit code: 1
  ╰─> [30 lines of output]
      running bdist_wheel
      running build
      running build_py
      creating build
      creating build/lib.macosx-11.0-arm64-cpython-311
      creating build/lib.macosx-11.0-arm64-cpython-311/RPi
      copying RPi/__init__.py -> build/lib.macosx-11.0-arm64-cpython-311/RPi
      creating build/lib.macosx-11.0-arm64-cpython-311/RPi/GPIO
      copying RPi/GPIO/__init__.py -> build/lib.macosx-11.0-arm64-cpython-311/RPi/GPIO
      running build_ext
      building 'RPi._GPIO' extension
      creating build/temp.macosx-11.0-arm64-cpython-311
      creating build/temp.macosx-11.0-arm64-cpython-311/source
      clang -DNDEBUG -fwrapv -O2 -Wall -fPIC -O2 -isystem /Users/ian/miniforge3/envs/gtk3/include -arch arm64 -fPIC -O2 -isystem /Users/ian/miniforge3/envs/gtk3/include -arch arm64 -I/Users/ian/miniforge3/envs/gtk3/include/python3.11 -c source/c_gpio.c -o build/temp.macosx-11.0-arm64-cpython-311/source/c_gpio.o
      source/c_gpio.c:170:9: warning: cast to smaller integer type 'uint32_t' (aka 'unsigned int') from 'uint8_t *' (aka 'unsigned char *') [-Wpointer-to-int-cast]
          if ((uint32_t)gpio_mem % PAGE_SIZE)
              ^~~~~~~~~~~~~~~~~~
      source/c_gpio.c:171:34: warning: cast to smaller integer type 'uint32_t' (aka 'unsigned int') from 'uint8_t *' (aka 'unsigned char *') [-Wpointer-to-int-cast]
              gpio_mem += PAGE_SIZE - ((uint32_t)gpio_mem % PAGE_SIZE);
                                       ^~~~~~~~~~~~~~~~~~
      2 warnings generated.
      clang -DNDEBUG -fwrapv -O2 -Wall -fPIC -O2 -isystem /Users/ian/miniforge3/envs/gtk3/include -arch arm64 -fPIC -O2 -isystem /Users/ian/miniforge3/envs/gtk3/include -arch arm64 -I/Users/ian/miniforge3/envs/gtk3/include/python3.11 -c source/common.c -o build/temp.macosx-11.0-arm64-cpython-311/source/common.o
      clang -DNDEBUG -fwrapv -O2 -Wall -fPIC -O2 -isystem /Users/ian/miniforge3/envs/gtk3/include -arch arm64 -fPIC -O2 -isystem /Users/ian/miniforge3/envs/gtk3/include -arch arm64 -I/Users/ian/miniforge3/envs/gtk3/include/python3.11 -c source/constants.c -o build/temp.macosx-11.0-arm64-cpython-311/source/constants.o
      clang -DNDEBUG -fwrapv -O2 -Wall -fPIC -O2 -isystem /Users/ian/miniforge3/envs/gtk3/include -arch arm64 -fPIC -O2 -isystem /Users/ian/miniforge3/envs/gtk3/include -arch arm64 -I/Users/ian/miniforge3/envs/gtk3/include/python3.11 -c source/cpuinfo.c -o build/temp.macosx-11.0-arm64-cpython-311/source/cpuinfo.o
      clang -DNDEBUG -fwrapv -O2 -Wall -fPIC -O2 -isystem /Users/ian/miniforge3/envs/gtk3/include -arch arm64 -fPIC -O2 -isystem /Users/ian/miniforge3/envs/gtk3/include -arch arm64 -I/Users/ian/miniforge3/envs/gtk3/include/python3.11 -c source/event_gpio.c -o build/temp.macosx-11.0-arm64-cpython-311/source/event_gpio.o
      source/event_gpio.c:24:10: fatal error: 'sys/epoll.h' file not found
      #include <sys/epoll.h>
               ^~~~~~~~~~~~~
      1 error generated.
      error: command '/usr/bin/clang' failed with exit code 1
      [end of output]
  
  note: This error originates from a subprocess, and is likely not a problem with pip.
  ERROR: Failed building wheel for RPi.GPIO
  Running setup.py clean for RPi.GPIO
Failed to build RPi.GPIO
Installing collected packages: RPi.GPIO, blinkt
  Running setup.py install for RPi.GPIO ... error
  error: subprocess-exited-with-error
  
  × Running setup.py install for RPi.GPIO did not run successfully.
  │ exit code: 1
  ╰─> [32 lines of output]
      running install
      /Users/ian/miniforge3/envs/gtk3/lib/python3.11/site-packages/setuptools/command/install.py:34: SetuptoolsDeprecationWarning: setup.py install is deprecated. Use build and pip and other standards-based tools.
        warnings.warn(
      running build
      running build_py
      creating build
      creating build/lib.macosx-11.0-arm64-cpython-311
      creating build/lib.macosx-11.0-arm64-cpython-311/RPi
      copying RPi/__init__.py -> build/lib.macosx-11.0-arm64-cpython-311/RPi
      creating build/lib.macosx-11.0-arm64-cpython-311/RPi/GPIO
      copying RPi/GPIO/__init__.py -> build/lib.macosx-11.0-arm64-cpython-311/RPi/GPIO
      running build_ext
      building 'RPi._GPIO' extension
      creating build/temp.macosx-11.0-arm64-cpython-311
      creating build/temp.macosx-11.0-arm64-cpython-311/source
      clang -DNDEBUG -fwrapv -O2 -Wall -fPIC -O2 -isystem /Users/ian/miniforge3/envs/gtk3/include -arch arm64 -fPIC -O2 -isystem /Users/ian/miniforge3/envs/gtk3/include -arch arm64 -I/Users/ian/miniforge3/envs/gtk3/include/python3.11 -c source/c_gpio.c -o build/temp.macosx-11.0-arm64-cpython-311/source/c_gpio.o
      source/c_gpio.c:170:9: warning: cast to smaller integer type 'uint32_t' (aka 'unsigned int') from 'uint8_t *' (aka 'unsigned char *') [-Wpointer-to-int-cast]
          if ((uint32_t)gpio_mem % PAGE_SIZE)
              ^~~~~~~~~~~~~~~~~~
      source/c_gpio.c:171:34: warning: cast to smaller integer type 'uint32_t' (aka 'unsigned int') from 'uint8_t *' (aka 'unsigned char *') [-Wpointer-to-int-cast]
              gpio_mem += PAGE_SIZE - ((uint32_t)gpio_mem % PAGE_SIZE);
                                       ^~~~~~~~~~~~~~~~~~
      2 warnings generated.
      clang -DNDEBUG -fwrapv -O2 -Wall -fPIC -O2 -isystem /Users/ian/miniforge3/envs/gtk3/include -arch arm64 -fPIC -O2 -isystem /Users/ian/miniforge3/envs/gtk3/include -arch arm64 -I/Users/ian/miniforge3/envs/gtk3/include/python3.11 -c source/common.c -o build/temp.macosx-11.0-arm64-cpython-311/source/common.o
      clang -DNDEBUG -fwrapv -O2 -Wall -fPIC -O2 -isystem /Users/ian/miniforge3/envs/gtk3/include -arch arm64 -fPIC -O2 -isystem /Users/ian/miniforge3/envs/gtk3/include -arch arm64 -I/Users/ian/miniforge3/envs/gtk3/include/python3.11 -c source/constants.c -o build/temp.macosx-11.0-arm64-cpython-311/source/constants.o
      clang -DNDEBUG -fwrapv -O2 -Wall -fPIC -O2 -isystem /Users/ian/miniforge3/envs/gtk3/include -arch arm64 -fPIC -O2 -isystem /Users/ian/miniforge3/envs/gtk3/include -arch arm64 -I/Users/ian/miniforge3/envs/gtk3/include/python3.11 -c source/cpuinfo.c -o build/temp.macosx-11.0-arm64-cpython-311/source/cpuinfo.o
      clang -DNDEBUG -fwrapv -O2 -Wall -fPIC -O2 -isystem /Users/ian/miniforge3/envs/gtk3/include -arch arm64 -fPIC -O2 -isystem /Users/ian/miniforge3/envs/gtk3/include -arch arm64 -I/Users/ian/miniforge3/envs/gtk3/include/python3.11 -c source/event_gpio.c -o build/temp.macosx-11.0-arm64-cpython-311/source/event_gpio.o
      source/event_gpio.c:24:10: fatal error: 'sys/epoll.h' file not found
      #include <sys/epoll.h>
               ^~~~~~~~~~~~~
      1 error generated.
      error: command '/usr/bin/clang' failed with exit code 1
      [end of output]
  
  note: This error originates from a subprocess, and is likely not a problem with pip.
error: legacy-install-failure

× Encountered error while trying to install package.
╰─> RPi.GPIO

note: This is an issue with the package mentioned above, not pip.
hint: See above for output from the failure.
(gtk3) /Users/ian/AAA/Code/self-driving-lab-demo/src $

sgbaird, 2023-01-10:
Hi @ianfoster, thanks for this! Good to know that `blinkt` doesn't install on MacOS. Thankfully, it's not needed for this project #164. The [`1.*`](https://github.com/sparks-baird/self-driving-lab-demo/tree/main/notebooks#1-blinkt-getting-started-deprecated-but-still-informative) and [`2.*`](https://github.com/sparks-baird/self-driving-lab-demo/tree/main/notebooks#2-search-algorithms-using-blinkt-deprecated-but-still-informative) notebooks use this, but are deprecated. There's still some useful content in those notebooks, but more for glancing over.

sgbaird, 2023-06-10:
These have been marked as deprecated. Closing for now, but lmk if you need anything else!

closed 2023-06-10 by sgbaird