sparks-baird/self-driving-lab-demo #194: Pico W setup with Ubuntu

stefanbringuier, 2023-05-04:
@sgbaird  I was final able to find time this weekend to assemble my kit and now I'm working on the micropython. I had to follow the instruction [here](https://www.raspberrypi.com/documentation/microcontrollers/micropython.html) since I couldn't see how to install the micropython firmware in the linux version of Thonny.  I  assigned the WiFi values in `secrets.py` and uploaded it to the Pico W but I'm getting the following issue:

```python
connected
ip = 192.168.10.206
Obtaining CA Certificate
Detected devices at I2C-addresses: 0x39
SD Card initialized successfully
Traceback (most recent call last):
  File "main.py", line 263, in <module>
  File "/lib/umqtt/simple.py", line 66, in connect
OSError: -16

Retrying client.connect() in 2 seconds...
Traceback (most recent call last):
  File "main.py", line 268, in <module>
  File "/lib/umqtt/simple.py", line 66, in connect
OSError: -16
```

Seems to be connection related. As a note, I set the MongoDB and HiveMQ env variables empty strings.

Any help is much appreciated. hoping to play around with this with my kids over the weekend. Thanks again for this awesome DYI project.

sgbaird, 2023-05-04:
> @sgbaird I was final able to find time this weekend to assemble my kit and now I'm working on the micropython. I had to follow the instruction [...]

Good to know about the Linux version! I'll make sure to add a note of that somewhere.

> I assigned the WiFi values in `secrets.py` and uploaded it to the Pico W but I'm getting the following issue:
>
> ```python
> connected
> ip = 192.168.10.206
> Obtaining CA Certificate
> Detected devices at [...]

The MongoDB variables can be safely set to empty strings; however, you'll need to use the default HiveMQ env variables (https://github.com/sparks-baird/self-driving-lab-demo/blob/src/public_mqtt_sdl_demo/sample_secrets.py#L12-L14) or create your own HiveMQ instance.

Originally I had the MQTT communication over a (free) public, unencrypted channel and later migrated that to a (free) private, encrypted channel. I shared the default authentication details publicly but created instructions for making your own instance in [these build instructions](https://github.com/sparks-baird/self-driving-lab-demo/blob/main/reports/CellPress_STARProtocols_clean2.docx) (stripped of [figures](https://github.com/sparks-baird/self-driving-lab-demo/tree/main/reports/figures/star-protocols) per journal submission requirements). The manuscript was recently accepted to Star Protocols 😁 I'll post the link to the article when it's available, and I'll update [the ChemRxiv preprint](https://chemrxiv.org/engage/chemrxiv/article-details/63b454cedadddc71b3a45c1f) soon.

I have some discussions around these topics at https://github.com/sparks-baird/self-driving-lab-demo/discussions/categories/data-and-access-management.

For my benefit, what resources were you looking at? At the time I made the YouTube video, I hadn't implemented this feature yet, and I don't think I updated the hackathon instructions either. I'll see if I can get a note added at the relevant point in the video, and I'll take a look at the hackathon instructions again.

> hoping to play around with this with my kids over the weekend. Thanks again for this awesome DYI project.

Awesome! I'm excited to hear how that goes, and you're welcome :) Thanks for giving this a shot!

sgbaird, 2023-05-04:
Lmk if you come up with any kid-friendly demos :) I picture fruit coming into play 🟠 🍌 🍓.

Also, since the retries for logging to MongoDB can add ~10-15 seconds between iterations (always maxes out on retries if MongoDB credentials are empty/incorrect), you may want to set [`mongodb=False`](https://self-driving-lab-demo.readthedocs.io/en/latest/api/self_driving_lab_demo.utils.html#self_driving_lab_demo.utils.observe.mqtt_observe_sensor_data) in [`observe_sensor_fn_kwargs`](https://self-driving-lab-demo.readthedocs.io/en/latest/api/self_driving_lab_demo.html#self_driving_lab_demo.core.SelfDrivingLabDemo).

Taking from the Colab tutorial, this would look like:
```python
from uuid import uuid4  # universally unique identifier
from self_driving_lab_demo import (
    SelfDrivingLabDemoLight,
    mqtt_observe_sensor_data,
    get_paho_client,
)

PICO_ID = "test"  # @param {type:"string"}
SESSION_ID = str(uuid4())  # random session ID
print(f"session ID: {SESSION_ID}")

# instantiate client once and reuse (to avoid opening too many connections)
client = get_paho_client(f"sdl-demo/picow/{PICO_ID}/as7341/")

sdl = SelfDrivingLabDemoLight(
    autoload=True,  # perform target data experiment automatically
    observe_sensor_data_fn=mqtt_observe_sensor_data,  # (default)
    observe_sensor_data_kwargs=dict(
        pico_id=PICO_ID, session_id=SESSION_ID, client=client, mongodb=False # <--- EDIT
    ),
)
```
Just gave it a try, and it seems ok 👍

stefanbringuier, 2023-05-05:
Thanks for the feedback! Will try things out.

> For my benefit, what resources were you looking at? 

I was mostly going through the README file and YouTube video for setup, but planning to go through the documentation preprint more closely now.

sgbaird, 2023-05-06:
Great, lmk! And thanks, I'll try to get the YouTube video up-to-date.