sparks-baird/self-driving-lab-demo #245: Certificate verification issue

viola-m-li, 2024-07-01:
Hi!

I'm having some issues setting up the software of the Pico W microcontroller. I tried to follow the given build instructions but I keep getting error messages related to the certificate verification:

```
connected
ip = 192.168.2.221
Obtaining CA Certificate
Detected devices at I2C-addresses: 0x39
Traceback (most recent call last):
  File "/lib/data_logging.py", line 56, in initialize_sdcard
  File "/lib/sdcard/sdcard.py", line 56, in __init__
  File "/lib/sdcard/sdcard.py", line 84, in init_card
OSError: no SD card

SD Card failed to initialize
Traceback (most recent call last):
  File "<stdin>", line 287, in <module>
  File "/lib/umqtt/simple.py", line 74, in connect
OSError: (-9984, 'MBEDTLS_ERR_X509_CERT_VERIFY_FAILED')

Retrying client.connect() in 2 seconds...
Traceback (most recent call last):
  File "<stdin>", line 292, in <module>
  File "/lib/umqtt/simple.py", line 74, in connect
OSError: (-9984, 'MBEDTLS_ERR_X509_CERT_VERIFY_FAILED')
```

I'm using the latest version of Thonny for MacOS (4.1.4 with Python 3.10) and the latest v0.8.8 sdl_demo.zip. For the MicroPython firmware I used the one that can be found in scripts/picow-setup/rp2-pico-w-20230309-unstable-v1.19.1-953-gb336b6bb7.uf2, which is dragged & dropped on to the RPI-RP2 drive.

Any help would be greatly appreciated. Thanks!

sgbaird, 2024-07-03:
Hi @viola-m-li, this is due to a recent change in how HiveMQ handles the certificates. The certificate for the default HiveMQ broker should be available in a new version `v0.8.9` shortly (as long as https://github.com/sparks-baird/self-driving-lab-demo/actions/runs/9779179700 completes successfully). However, if you create your own HiveMQ broker, you will need to generate your own via the instructions in the warning at https://ac-microcourses.readthedocs.io/en/latest/courses/hello-world/1.1-running-the-demo.html, copied here for convenience:

> Recently, HiveMQ Cloud changed such that `hivemq-com-chain.der` (a Certificate Authority (CA) file) is not transferrable across different broker instances. The [latest `hivemq-com-chain.der` file](https://raw.githubusercontent.com/sparks-baird/self-driving-lab-demo/main/src/public_mqtt_sdl_demo/hivemq-com-chain.der) from [`self-driving-lab-demo`](https://github.com/sparks-baird/self-driving-lab-demo) will be hard-coded to the `self-driving-lab-demo` public test credentials (i.e., what is used in Module 1 - Running the Demo), so the *tutorials* should run without issue as long as you are using that file. However, the *assignment* requires you to have your own HiveMQ Cloud broker instance, so you will need to [generate a `hivemq-com-chain.der` file specific to your instance](https://colab.research.google.com/github/sparks-baird/self-driving-lab-demo/blob/main/notebooks/7.2.1-hivemq-openssl-certificate.ipynb) and upload it to your microcontroller in place of the default one.

Do you have any suggestions for where this would best be clarified in the `self-driving-lab-demo` repo? For example:
- Add a note in the README
- Add the warning from above to https://colab.research.google.com/github/sparks-baird/self-driving-lab-demo/blob/main/notebooks/4.2-paho-mqtt-colab-sdl-demo-test.ipynb
- Update the tutorial video

The manuscript is the hardest place to change this - I've requested a minor fix to a link before, but I don't think I've heard back from them after months. This is pretty specific to the journal.

viola-m-li, 2024-07-04:
Hi @sgbaird, thanks for the clarification! It works now.

Having a more useful error would have been the most helpful for me since this is the first thing you see and try to fix before going to the notebook. 
A section/note in the README that provides some updated information on changes since the release of the tutorial video (/manuscript) might also be a good idea.

sgbaird, 2024-07-04:
@viola-m-li thanks! I updated main.py. Do you mind testing this? https://github.com/sparks-baird/self-driving-lab-demo/blob/hivemq/src/public_mqtt_sdl_demo/main.py

viola-m-li, 2024-07-05:
@sgbaird I tested it and it look good for me :)

closed 2024-09-21 by sgbaird

jsmz2602, 2025-06-03:
@sgbaird found the solution and also found out that this issue was discussed already [here](https://gist.github.com/sgbaird/5ddef425e8d4aae454a69fbce8654faf)