sparks-baird/self-driving-lab-demo #252: [1.1 Run the Demo] Cannot get SDL demo test to run successfully

uznab16, 2024-10-15:
I'm trying to run the demo test with the LED colors (4.2-paho...), but I run into errors after installing the Python package. I'm not sure how to resolve it as I followed all of the instructions before this.
[image]

sgbaird, 2024-10-17:
@uznab16 can you include the full stack trace as text? Hard to see what's going on based on what you shared.

sgbaird, 2024-10-17:
Also, the public demo was offline for a bit. Maybe try again?

uznab16, 2024-10-18:
---------------------------------------------------------------------------
Empty                                     Traceback (most recent call last)
[/usr/local/lib/python3.10/dist-packages/self_driving_lab_demo/utils/observe.py](https://localhost:8080/#) in mqtt_observe_sensor_data(R, G, B, atime, astep, gain, pico_id, session_id, timeout, queue_timeout, client, username, password, hostname, port, tls, mongodb, extra_info)
    105         try:
--> 106             sensor_data = sensor_data_queue.get(True, queue_timeout)
    107         except Empty as e:

6 frames
[/usr/lib/python3.10/queue.py](https://localhost:8080/#) in get(self, block, timeout)
    178                     if remaining <= 0.0:
--> 179                         raise Empty
    180                     self.not_empty.wait(remaining)

Empty: 

The above exception was the direct cause of the following exception:

Empty                                     Traceback (most recent call last)
[<ipython-input-15-973ce92dfadf>](https://localhost:8080/#) in <cell line: 31>()
     29 )
     30 
---> 31 sdl = SelfDrivingLabDemoLight(
     32     autoload=True,  # perform target data experiment automatically
     33     target_inputs=target_inputs,  # if None, then defaults to random color using `target_seed` attribute

[/usr/local/lib/python3.10/dist-packages/self_driving_lab_demo/demos/light.py](https://localhost:8080/#) in __init__(self, random_rng, target_seed, target_inputs, rest_seconds, max_power, autoload, simulation, simulator, observe_sensor_data_fn, observe_sensor_data_kwargs)
    158         >>> demo.evaluate(dict(R=50, G=150, B=250))
    159         """
--> 160         super().__init__(
    161             random_rng=random_rng,
    162             target_seed=target_seed,

[/usr/local/lib/python3.10/dist-packages/self_driving_lab_demo/core.py](https://localhost:8080/#) in __init__(self, random_rng, target_seed, target_inputs, rest_seconds, max_power, autoload, simulation, simulator, observe_sensor_data_fn, observe_sensor_data_kwargs)
     75         if autoload:
     76             # must come after creating sensor attribute
---> 77             self.observe_target_results()
     78 
     79     def observe_sensor_data(self, parameters):

[/usr/local/lib/python3.10/dist-packages/self_driving_lab_demo/core.py](https://localhost:8080/#) in observe_target_results(self)
    107 
    108     def observe_target_results(self):
--> 109         self.target_results = self.observe_sensor_data(self.get_target_inputs())
    110         if self.target_results is None or (
    111             self.target_results.get("error", None) is not None

[/usr/local/lib/python3.10/dist-packages/self_driving_lab_demo/core.py](https://localhost:8080/#) in observe_sensor_data(self, parameters)
     82         sleep(self.rest_seconds)
     83 
---> 84         return self.observe_sensor_data_fn(
     85             **parameters, **self.observe_sensor_data_kwargs
     86         )

[/usr/local/lib/python3.10/dist-packages/self_driving_lab_demo/utils/observe.py](https://localhost:8080/#) in mqtt_observe_sensor_data(R, G, B, atime, astep, gain, pico_id, session_id, timeout, queue_timeout, client, username, password, hostname, port, tls, mongodb, extra_info)
    106             sensor_data = sensor_data_queue.get(True, queue_timeout)
    107         except Empty as e:
--> 108             raise Empty(
    109                 f"Sensor data retrieval timed out ({queue_timeout} seconds)"
    110             ) from e

Empty: Sensor data retrieval timed out (60 seconds)

Lopeznil, 2024-10-25:
I am getting the same exact error

sgbaird, 2024-10-31:
Could you try one more time? We've been having a bunch of issues with certificates, but I think we have a fix for these.

I just verified with https://huggingface.co/spaces/AccelerationConsortium/light-mixing, but maybe it's something specific to the notebook, in which case I'll reproduce.

Can you also verify that it works correctly if you set `simulation=True`?

Sorry for the trouble with this. Also, I've updated my notifications for this repo so hopefully I see these sooner.

sgbaird, 2024-10-31:
I see that the same error is occurring in the unit tests: https://github.com/sparks-baird/self-driving-lab-demo/actions/runs/11602495968/job/32307659409#step:5:1.

Since the device appears to be online and accessible through the hugging face spaces app, I'll need to debug a bit more to figure out the source of the issue.

sgbaird, 2024-10-31:
I just ran parts of https://colab.research.google.com/github/sparks-baird/self-driving-lab-demo/blob/main/notebooks/4.2-paho-mqtt-colab-sdl-demo-test.ipynb#scrollTo=N88u4HkItJoj and it seems to be ok.

[image]

Are you doing this with your own microcontroller, or are you trying to access the public test demo?

uznab16, 2024-10-31:
@sgbaird I tried to run the microcontroller (I have my own) one more time and still got the same error. I then altered the code to make simulation=True (line 36) and the code compiled, BUT the LED light did not turn on or turn purple-ish.

Leaving simulation=True and letting it compile, I ran the sdl.observe_sensor_data code and although it compiled, the LED did not turn on. Am I doing something wrong?

sgbaird, 2024-10-31:
@uznab16 maybe have a look at https://ac-microcourses.readthedocs.io/en/latest/courses/hello-world/1.1-running-the-demo.html.

Can you copy your output from Thonny or wherever you're running the microcontroller script and paste it here?

Are you trying this with the default public test broker or your own broker?

Due to some recent changes, you'll also need to generate your own certificate (see instructions in linked microcourse page).

uznab16, 2024-10-31:
**@sgbaird This is the feedback I get after running main.py in Thonny**:
------------------------------------------------------------------------------
>>> %Run -c $EDITOR_CONTENT

MPY: soft reboot
RSA parameters (keep private):
{'e': 65537, 'bits': 256, 'd': 60149547603089246402076931315474481926489343369511888292782728727795363801233, 'n': 66119102671312645780827169534086734687139341511602084818941201465477938020457}

PICO_ID: ###

Unencrypted PICO ID (keep private): ###
Encrypted PICO ID (OK to share publicly): 38602889212947404908625381720892202114024157235484360548200632962736012813675
Truncated, encrypted PICO ID (OK to share publicly): 3860288921
MQTT prefix: sdl-demo/picow/e6632c8593733f30/
MAC address: d8:3a:dd:fe:99:8d
connected
ip = 10.51.0.202
-----------------------------------------------------------------------------

**However, a few minutes later, it will show this**:
-----------------------------------------------------------------------------
Traceback (most recent call last):
  File "<stdin>", line 99, in <module>
  File "ntptime.py", line 1, in settime
  File "ntptime.py", line 1, in time
OSError: [Errno 110] ETIMEDOUT

Retrying ntptime.settime(), still with pool.ntp.org
Traceback (most recent call last):
  File "<stdin>", line 104, in <module>
  File "ntptime.py", line 1, in settime
  File "ntptime.py", line 1, in time
OSError: [Errno 110] ETIMEDOUT

Retrying ntptime.settime() with time.google.com
Traceback (most recent call last):
  File "<stdin>", line 109, in <module>
  File "ntptime.py", line 1, in settime
  File "ntptime.py", line 1, in time
OSError: [Errno 110] ETIMEDOUT

PROBLEM IN THONNY'S BACK-END: Exception while handling 'Run' (ConnectionError: EOF).
See Thonny's backend.log for more info.
You may need to press "Stop/Restart" or hard-reset your MicroPython device and try again.

Process ended with exit code 1.
────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────
Traceback (most recent call last):
  File "main.py", line 24, in <module>
KeyboardInterrupt: 
MicroPython v1.23.0 on 2024-06-02; Raspberry Pi Pico W with RP2040
Type "help()" for more information.

MPY: soft reboot
MicroPython v1.23.0 on 2024-06-02; Raspberry Pi Pico W with RP2040

Type "help()" for more information.

>>> 
Process ended with exit code 3221225786.
-------------------------------------------------------------------------------

**The actual pico_ID is listed. I just want to post it here. I'm sorry; I don't know what you mean by broker. I have the hivemq-com-chain.pem file from the microcourse included in the sdl_demo folder and I highlighted it as well in the "Upload to /" step. Just now, I have to create my own certificate by creating a HiveMQ account followed by a Serverless Cluster. Then I used the URL from the cluster as my host name to run the code in 7.2.1-hivemq-openssl-certificate.ipynb. I then uploaded the certificate to the Pico and reran main.py in Thonny. The output is generally the same as above. Sometimes I do have to press the "Stop" button. Other times, I get this:**

Traceback (most recent call last):
  File "main.py", line 44, in <module>
KeyboardInterrupt: 
MicroPython v1.23.0 on 2024-06-02; Raspberry Pi Pico W with RP2040
Type "help()" for more information.

MPY: soft reboot

sgbaird, 2024-11-01:
If you can reproducible get the following error

```python
Traceback (most recent call last):
File "", line 99, in
File "ntptime.py", line 1, in settime
File "ntptime.py", line 1, in time
OSError: [Errno 110] ETIMEDOUT
```

and you've been able to connect to the WiFi as you showed, then it seems likely it's an issue with the WiFi you're using.

Could you comment a bit on that setup? Are you able to test with any other WiFi's for debugging?

uznab16, 2024-11-04:
I'm using my company WiFi which may present some security issues. I will try and Hot Spot my personal phone to see if it solves the issue.

uznab16, 2024-11-04:
@sgbaird Using my phone's hot spot has solved the issue. I'm able to run the code and see the LED turn on. Thank you, Sterling!