sparks-baird/self-driving-lab-demo #255: "Monster Blood" CLSLab: Liquid Motor does not respond to Pi

kinstonwithoutg, 2024-11-01:
The Pi seems to be getting commands as the lights blink when we start the `Instantiation` on Google Colab, and the PI seems to be working and waiting for experiment requests. Still, the motors are not doing anything, which resulted in the sensor running timeout.  We tried different boards, and the problem still exists.  

It was later found that the issue is not on the board as all boards or networks face the same issue - it should be a software one.

sgbaird, 2024-11-02:
Could you include a picture of the setup? Also, please copy the main.py file (or include the link to the one you're using). There have been some weird things with certificates lately (will likely have a fix in the coming month) #253, so I recommend [regenerating your certificate](https://colab.research.google.com/github/sparks-baird/self-driving-lab-demo/blob/main/notebooks/7.2.1-hivemq-openssl-certificate.ipynb#scrollTo=c3a0_l6qtwZc). Are you using a custom broker?

Also, can you verify whether or not you can use https://huggingface.co/spaces/AccelerationConsortium/light-mixing with a light-mixing demo that you have on hand?

sgbaird, 2024-11-02:
Some other random bits present in https://ac-microcourses.readthedocs.io/en/latest/courses/hello-world/1.1-running-the-demo.html

It sounds like you've already tried multiple networks (e.g., phone hotspot).

kinstonwithoutg, 2024-11-02:
[image]
We are using the U's IoT wifi with multiple different connection parties, we will try to regenerate the certificate and with the light demo tomorrow.   

`"""Run a materials acceleration platform demo on a Raspberry Pi Pico W."""

import gc
try:
    import json
except ImportError:
    import simplejson as json
import json
import os
import ssl
from secrets import HIVEMQ_HOST, HIVEMQ_PASSWORD, HIVEMQ_USERNAME, PASSWORD, SSID
from time import sleep

import ntptime
from as7341_sensor import Sensor
from data_logging import initialize_sdcard, log_to_mongodb
from machine import PWM, Pin, reset, unique_id
from netman import connectWiFi
from sdl_demo_utils import (
    Experiment,
    encrypt_id,
    get_onboard_led,
    get_traceback,
    heartbeat,
    sign_of_life,
)
from ubinascii import hexlify
from umqtt.simple import MQTTClient

try:
    from secrets import (
        DEVICE_NICKNAME,
        MONGODB_API_KEY,
        MONGODB_APP_NAME,
        MONGODB_CLUSTER_NAME,
        MONGODB_COLLECTION_NAME,
        MONGODB_DATABASE_NAME,
    )
except Exception as e:
    print(get_traceback(e))

##### BEGIN USER-DEFINED IMPORTS #####
from time import ticks_diff, ticks_ms

##### END USER-DEFINED IMPORTS #####

# sleep to avoid KeyboardInterrupt overwriting log.txt when opening in Thonny
sleep(5.0)

try:
    gc.collect()
    port = 8883

    logfile = open("log.txt", "w")
    # duplicate stdout and stderr to the log file
    os.dupterm(logfile)

    # https://medium.com/@johnlpage/introduction-to-microcontrollers-and-the-pi-pico-w-f7a2d9ad1394
    # you can request a MongoDB collection specific to you by emailing
    # sterling.baird@utah.edu
    mongodb_url = f"https://data.mongodb-api.com/app/{MONGODB_APP_NAME}/endpoint/data/v1/action/insertOne"  # noqa: E501

    my_id = hexlify(unique_id()).decode()
    my_encrypted_id = encrypt_id(my_id, verbose=True)

    # # aside: for sgbaird's public test demo only
    # my_id = "test"
[... 656 lines cut ...]
            wdt.feed()  # Feed the watchdog timer to prevent reset
        except Exception as e:
            print(get_traceback(e))
            print("Reconnecting to WiFi...")
            connectWiFi(SSID, PASSWORD, country="US")
            client.connect(clean_session=False)
            client.set_callback(callback)
            client.subscribe(prefix + "GPIO/#")
            wdt.feed()  # Ensure watchdog is fed even after reconnecting
except Exception as e:
    print(get_traceback(e))
    fname = "error.txt"
    logfile = open(fname, "w")
    logfile.write(get_traceback(e))
    logfile.close()

# %% Code Graveyard

#             logfile = open("log.txt", "a")
#             logfile.write(get_traceback(e))
#             os.dupterm(logfile)
#             logfile.close()
#             os.dupterm(None)
`
The first one was used in September, but it does not work anymore.  The second one was downloaded yesterday but still does not work.  (I have not check the difference yet)

kinstonwithoutg, 2024-11-02:
I regenerated the certificate, but the issue still persists.  I have also tried the light demo, and the sensor error still pops up, I guess it is the sensor problem rather than motors?

it seems like the main.py can't find the as7341_sensor module as well.

sgbaird, 2024-11-04:
What is the output from both the notebook and from the microcontroller when you're running it?

Have you tried connecting to a mobile hotspot in extended compatibility mode?

> Could you include a picture of the setup? Also, please copy the main.py file (or include the link to the one you're using). There have been some weird things with certificates lately (will likely have a fix in the coming month) #253, so I recommend [regenerating your certificate](https://colab.research.google.com/github/sparks-baird/self-driving-lab-demo/blob/main/notebooks/7.2.1-hivemq-openssl-certificate.ipynb#scrollTo=c3a0_l6qtwZc). Are you using a custom broker?
> 
> Also, can you verify whether or not you can use [huggingface.co/spaces/AccelerationConsortium/light-mixing](https://huggingface.co/spaces/AccelerationConsortium/light-mixing) with a light-mixing demo that you have on hand?

Can you clarify the other questions?

sgbaird, 2025-01-06:
@kinstonwithoutg any resolution to the problem?

kinstonwithoutg, 2025-01-23:
The notebook was throwing a runtime error, and turns out the microcontroller couldn’t locate the AS7341 sensor module. Based on the infrastructure my team built for running experimental campaigns in our research, we’ve decided to overhaul the entire main.py to work around this.

closed 2025-01-23 by kinstonwithoutg

sgbaird, 2025-01-23:
Nice! For reference, do you have your version of main.py that you could share? (E.g., GitHub link). Or you could post it directly here if that's a pain.