sparks-baird/self-driving-lab-demo #256: Autograder not working

cjbrew, 2024-11-12:
Hello,

I was able to use the autograder module for the orientation, but now it's not working for the demo. Github gives me the following error: 

Error running command github-actions.explorer.triggerRun: Cannot read properties of undefined (reading 'workflow_dispatch'). This is likely caused by the extension that contributes github-actions.explorer.triggerRun.

In Line 4 of the workflow file, Github seems to be having trouble with the dispatch.

If this issue sounds familiar, I'd appreciate any help you can offer!

closed 2024-11-18 by cjbrew

sgbaird, 2024-11-19:
Sorry for the lack of response. Were you able to resolve?

cjbrew, 2024-11-19:
Yes; I had to merge a new branch I'd made and manually sync the committed changes.