sparks-baird/self-driving-lab-demo #261: Atlas Data API and HTTPS Endpoints Deprecation

jsmz2602, 2025-03-21:
Atlas Data API, a fully-managed middleware service, [deprecated](https://www.mongodb.com/docs/atlas/app-services/data-api/data-api-deprecation/) on September 30, 2025, which impacts data logging in our current setting for SDL (Lightmixing) demo.

Is there any alternative solution yet?

jsmz2602, 2025-03-21:
Only feasible (= pythonic & free) solution for me among all options listed would be via [Python and FastAPI](https://www.mongodb.com/docs/atlas/app-services/data-api/data-api-deprecation/#python-and-fastapi).

@sgbaird

sgbaird, 2025-03-21:
We've switched to Lambda functions: https://ac-microcourses.readthedocs.io/en/latest/courses/hello-world/1.5-data-logging.html (+ https://ac-microcourses.readthedocs.io/en/latest/courses/hello-world/1.5.1-pymongo.html on orchestrator side), but I haven't incorporated that into `self-driving-lab-demo` yet. Shoot me an email at the one listed at https://github.com/sgbaird and I'll send you an assignment from that course that has detailed instructions for the setup.

sgbaird, 2025-03-21:
Here is a link to some instructions you can follow to get it set up: https://github.com/ACC-HelloWorld/5-data-logging

Would it help to get this incorporated into `self-driving-lab-demo` soon?