sparks-baird/self-driving-lab-demo #263: import error on main.py -> umqtt/simple.py

kinstonwithoutg, 2025-04-01:
[image]

it seems like we remember to change ussl into ssl on the main but not on the umqtt/simple.py

kinstonwithoutg, 2025-04-01:
[image]

sgbaird, 2025-04-01:
Ah, yeah I've almost entirely shifted to mqtt_as for future development. There was a discrepancy between MicroPython versions I think, too. What version are you running of MicroPython?

https://github.com/AccelerationConsortium/ac-microcourses/issues/96

(mqtt_as example: https://ac-microcourses.readthedocs.io/en/latest/courses/hello-world/1.4-hardware-software-communication.html)

kinstonwithoutg, 2025-04-01:
We are running 1.24.1 MicroPython, how far back we should go back to make it working?

sgbaird, 2025-04-01:
Would you be able to make a PR for simple? https://github.com/AccelerationConsortium/ac-microcourses/issues/96#issuecomment-2411593221

kinstonwithoutg, 2025-04-01:
I think the simple.py on the main branch is up to date, but when I import the main.py and secret.py on the Pi, it was calling the old version (?

kinstonwithoutg, 2025-04-01:
[image]
Good old wifi certificate issue now oof

closed 2025-04-01 by kinstonwithoutg

sgbaird, 2025-04-01:
Oh no 😢 I probably need to either stop hardcoding the certificate at all and telling people they need to generate it (I assume you're using the colab notebook from this repo?) or set up a GitHub action to autogenerate it.

sgbaird, 2025-04-01:
Not sure why you'd need to generate so many. But it might help with sanity to delete the certificate completely before uploading the new one. I remember having some issues related to that, vaguely. Or maybe dejavu

kinstonwithoutg, 2025-04-01:
I used the colab notebook to regenerate, but the first three did not work, so I generated a few, and eventually they work (just not sure how long)

sgbaird, 2025-04-01:
Wow, strange. I'll try to keep an eye out for that. I think the end of the notebook now has a command that shows the expiration date for the certificate.